ispapp/ispapp-routeros-client #3: Script not working on Cloud Hosted Router

sundialcommunications, 2021-07-08:
Error when importing script: "input does not match any value of value-name"
Host does not appear in Unknown Hosts.

[image]

andrewhodel, 2021-07-08:
Email me access credentials for ssh and winbox referencing this issue at ***@***.***

Thank You,
Andrew Hodel

andrewhodel, 2021-07-09:
Is this the same wan ip problem? @sundialcommunications

andrewhodel, 2021-07-10:
You didn't allow the collectors script to run before the cmd script, you didn't download the raw file and edit it or something it works fine.

[image]

closed 2021-07-10 by andrewhodel