ispapp/ispapp-routeros-client #4: WAN IP Error

mthoodlum, 2021-07-09:
Host shows offline on the demo server even though it is actually online. 
[image]

andrewhodel, 2021-07-09:
This has nothing to do with a wan ip error. 

The host shows online with the proper credentials.

andrewhodel, 2021-07-09:
Perhaps it is not sending an empty string when it cannot find a wan ip. 

How does it get a default route?

mthoodlum, 2021-07-09:
The WAN interface IP address is the IP address associated with the interface that has the 0.0.0.0/0 default route that is "active". With MikroTik you can have multiple default routes but only one of them will be active. MikroTik can also have multiple IP addresses on a single interface so this complicates if further. We never got the logic quite right.

andrewhodel, 2021-08-23:
fixed in version .29 https://github.com/ispapp/ispapp-routeros-client/commit/bab8c2f0cba2c6b27cc38f7b003d08d79e883a86

closed 2021-08-23 by andrewhodel