ispapp/ispapp-routeros-client #10: Special characters in key does not work

mthoodlum, 2021-07-16:
I tried to use a key with special characters and the key did not work. The key that did not work was this:

G$4i0*nwKKl!Ow%CnrIWHE3YzrJR0J9@7qbtyyq4n5g

Changing it to a key with only letters and numbers fixed the problem.

andrewhodel, 2021-11-19:
The `$` character in routeros, and a few others need to be escaped.

```
\" | Insert double quote
-- | --
\\ | Insert backslash
\n | Insert newline
\r | Insert carriage return
\t | Insert horizontal tab
\$ | Output $ character. Otherwise $ is used to link variable.
\? | Output ? character. Otherwise ? is used to print "help" in console.
\_ | - space
\a | - BEL (0x07)
\b | - backspace (0x08)
\f | - form feed (0xFF)
\v | Insert vertical tab
\xx | Print character from hex value. Hex number should use capital letters.
``

andrewhodel, 2021-11-19:
Tested, only these characters need to be escaped with version 6.48.4

```
$
```

If you put a `?` or `\?` in the key string it returns as a string with only a `?`.

The server will only accept keys without `$` or `\` and this will be solved for the ISPApp routeros client.

closed 2021-11-19 by andrewhodel