ispapp/ispapp-routeros-client #15: Invalid characters in the DHCP leases table causes RouterOS client script to fail

mthoodlum, 2021-08-29:
[image]

mthoodlum, 2021-08-29:
This is still the same problem as #11 and #14

mthoodlum, 2021-08-29:
I removed the block of Emrah from the ISPApp organization.

andrewhodel, 2021-08-29:
Why are you repeating this. 

We had this discussion, you submitted a bug to mikrotik and we are waiting on them to issue a fix. 

They are not following the dhcp protocol.

mthoodlum, 2021-08-29:
We can't wait for MikroTik. They may never fix this problem and we need this fixed to launch. We need to sanitize the input so the RouterOS client scripts don't fail because of invalid data.

andrewhodel, 2021-08-29:
No, Mikrotik must adhere to the DHCP protocol.

It's absurd to ruin code with bloat because of someone not doing something right.

mthoodlum, 2021-08-29:
Sanitizing input isn't bloat. We should be doing this anyway.

mthoodlum, 2021-08-29:
It is also absurd to wait on MikroTik to launch. We need to launch.

mthoodlum, 2021-08-29:
https://en.wikipedia.org/wiki/Improper_input_validation

andrewhodel, 2021-08-29:
The improper input validation is the dhcp server not validating the input from the dhcp client!

We are not going to validate inputs from protocols!  That leads to bloated code.

Why don't we validate rssi levels and argue forever about what the minimum is and how long the float should be when it's already a number!

closed 2021-08-29 by andrewhodel