ispapp/ispapp-routeros-client #17: v1.10 sending uppercase characters in login

mthoodlum, 2021-09-05:
There needs to be consistency in case when the device sends its login info. Historically we have always used lowercase so we need to continue only using lowercase. The v1.10 script sends the uppercase Wi-Fi MAC address as the login info as shown in this screenshot.
[image]

mthoodlum, 2021-09-05:
[image]

andrewhodel, 2021-09-05:
That is not true, you can see in the json string that it is being sent as lowercase.

mthoodlum, 2021-09-05:
Uppercase characters are shown before lowercase characters are shown. Either the client script is sending uppercase characters or the server is interpreting them as uppercase characters. Either way they show up in the Unknown Hosts as uppercase characters are per the screenshot.

andrewhodel, 2021-09-05:
> Uppercase characters are shown before lowercase characters are shown. Either the client script is sending uppercase characters or the server is interpreting them as uppercase characters. Either way they show up in the Unknown Hosts as uppercase characters are per the screenshot.

Read the code David, the mac address is gathered directly from the interface and stored in a variable that is then converted to lower case.  It is printed as gathered, then again after being moved to lower case.

The request is then made with the variable and the json is printed out showing that it is being sent lowercase.

I cannot help it that someone you shared the key with made a request with uppercase characters.

closed 2021-09-05 by andrewhodel