ispapp/ispapp-routeros-client #18: lte info without api

andrewhodel, 2021-10-17:
```
/interface lte info 0 once
```

Parseable at least, and better than using the tcp API within the scripting language.

Keep in mind that it does not always return all the data as if it was connected like fields would.

```
/interface lte info 0 once  
     pin-status: ok
  functionality: full
   manufacturer: "MikroTik"
          model: "R11e-LTE6"
       revision: R11e-LTE6_V028
           imei: 6754765
           imsi: 567
           uicc: 567567
```

andrewhodel, 2021-11-19:
fixed https://github.com/ispapp/ispapp-routeros-client/commit/bbcdfef2addb851ccb9c9951956c1aa8741bb688

closed 2021-11-19 by andrewhodel

mthoodlum, 2021-11-23:
[image]