ispapp/ispapp-routeros-client #19: WAN interface IP address

mthoodlum, 2021-11-02:
This continues to be a problem. The problematic function is commented with "#WAN Port IP Address" on line 1425: https://github.com/ispapp/ispapp-routeros-client/blob/master/ispapp.rsc#L1425

The variable "interfaceIp" is actually the "gatewayIp" and should be changed.

Proper logic needs to take the gateway IP address (the IP address closest to the interface name in the "gateway-status" variable) and find the corresponding network in the IP addresses. To do this properly we need to do a subnet calculation and determine which subnet includes this gateway IP address. A subnet calculation can be performed with a bitwise calculation:

https://wiki.mikrotik.com/wiki/Manual:Scripting#Bitwise_Operators

andrewhodel, 2021-11-02:
No you don't David.

The IP Address is a string with a subnet in IPv4 and in IPv6 it is another menu item `/ipv6 ...`.

You split the string by the `/` character, then you have the IP address as the first element in the array.

There's nothing else to it.

mthoodlum, 2021-11-02:
You have to loop through the IP addresses to match the IP address with the gateway.

andrewhodel, 2021-11-02:
No David.

You do not.  The logic is correct.

You get the interface for the default route, then the ip address of that interface and split the string of that (the ip address) by / and take the first string element of the string.

Stop being stupid about it, you don't do anything with bitwise calculations to split a string.

mthoodlum, 2021-11-02:
[image]

andrewhodel, 2021-11-02:
Exactly, and the address is a string that you need to split by `/`.

mthoodlum, 2021-11-02:
The IP address in the IP routes is not the IP address of the interface, it is the IP address of the gateway. You need to associate the gateway address with the interface address. You can't just "get 0" where name=Ether6 because there are two IP addresses associated with Ether6.

andrewhodel, 2021-11-02:
David,

READ THIS:

You get the interface for the default route, then the ip address of that interface and split the string of that (the ip address) by / and take the first string element of the string.

1. interface for the default route
2. ip address of that interface
3. split that string by / and take the first string element of the string

mthoodlum, 2021-11-02:
[image]

mthoodlum, 2021-11-02:
[image]

andrewhodel, 2021-11-02:
1. interface for the default route (line 54 sets that to gatewayStatus)
2. ip address of that interface (line 63 sets that to interfaceIp from the value in the index gathered on line 61 and set to getInterfaceIndex)
3. split that string by / and take the first string element of the string (no string match in winbox vs that returned by the index, so get it from gatewayStatus split by space on line 61 then find the matching IP as there is no possibility of another IP with the same network on a router via line 72; set to wanIp)

You are wrong David, it is done.

andrewhodel, 2021-11-02:
Line 72 finds it based on the string split by space regardless, you are not understanding the third argument to find being set to -1 that means start, -1 doesn't mean anything as it is lower than the start.

What you are doing is absurd.

[image]

andrewhodel, 2021-11-02:
It's exactly what Mikrotik's documentation says to do:

https://wiki.mikrotik.com/wiki/Manual:Scripting

I don't know why you can't understand these things.

andrewhodel, 2021-11-02:
Here's a data dump of their HTML with a timestamp showing this.

[scripting-routeros.html.txt](https://github.com/ispapp/ispapp-routeros-client/files/7457450/scripting-routeros.html.txt)

mthoodlum, 2021-11-02:
Line 66 removes a character from the end of the string that is supposed to be the network address.
Line 67 adds a "0" to the previously shortened string.

andrewhodel, 2021-11-02:
Then it's obvious that Emrah whom you paid too little to build something wrote a problem into the script to only happen with certain IP ranges.

Why would you continue the lie by trying to say use bitwise calculations?

This is why I want to document this, it's evil to do to people.

andrewhodel, 2021-11-02:
I don't know if line 66 does what you say, that's not relevant when you are saying these things.

It's very simple logic, why pay someone to write broken routines?

mthoodlum, 2021-11-02:
Because there are multiple IP addresses assigned to interfaces. You can't assume the IP address attached to the WAN port is the correct address.

andrewhodel, 2021-11-02:
What are you talking about David, that's why you get the default route and match the network address of it!

andrewhodel, 2021-11-02:
If a default route (that of destination 0.0.0.0/0) has a network address, that address will match the IP address' network address for the interface.  It won't be able to have two IP addresses on the same network and the same interface!

Why don't you understand these things?

andrewhodel, 2021-11-02:
I don't understand why you aren't doing this to billionaires with every pen they use.

mthoodlum, 2021-11-02:
The default route doesn't have a network address. It has a gateway address.

andrewhodel, 2021-11-02:
David you have to stop lying, this code in your screenshot is not the same code.  You are putting the extra 0 it isn't there.

[image]

andrewhodel, 2021-11-02:
Do you not understand it is replacing the 1 with a 0 to get the network address? The network address is always a 0 at the end for a /24 and 0.0 for a /16 and 0.0.0 for a /8

It should be split by the periods if anything but the find command does not work as documented in SUP-61666.

mthoodlum, 2021-11-02:
"set" sets variable to that value. "put" prints the value to the screen. I added the "put" lines to make the values visible.

andrewhodel, 2021-11-02:
David, it works on a /16 do you not understand that?

andrewhodel, 2021-11-02:
It works on a /24 also.

It doesn't work on a router you are working on.

There's no problem, you aren't printing ipNetworkAddress and intentionally trying to confuse.

STOP LYING.

mthoodlum, 2021-11-02:
No it doesn't. it only works if the gateway happens to be one of the first 9 addresses in a /16. If the gateway is the last host address, like Comcast does it, then it will never work.

andrewhodel, 2021-11-02:
That's not relevant David, you lied about the setup on the router you are using it on and it isn't a real problem.

You paid Emrah to write problems.

When it's a real customer issue, I'll fix it.

andrewhodel, 2021-11-02:
You aren't testing, you are intentionally paying people to write broken things to ruin people that really work.

andrewhodel, 2021-11-02:
It won't crash the router, it won't stop the Internet connection and things will be done without you blocking access to and changing things to intentionally try to confuse.

There's no reason to waste this much time, that's all there is to it.

mthoodlum, 2021-11-02:
Devices report offline that are actually online, like the hAPs on the HowTo network. It is a known bug and I am reopening the issue.

andrewhodel, 2021-11-02:
No it is not David, it is like you getting in a car and driving into someone then saying it's a known bug that cars can drive into someone.

You are trying to maintain a linear story and not respecting the idea of lidar and no-lidar.

andrewhodel, 2021-11-02:
You intentionally broke it and created confusion after paying Emrah to create confusion in the first place then you were confused enough to document how.  Just stop.

andrewhodel, 2021-11-02:
You are letting people that lie about past stories manipulate them slightly then get you to follow the logic and lie about something for you to try to understand something.

andrewhodel, 2021-11-02:
I hate that you come to me and lie about things the state and legislators who are absolute millionaires can't figure out and ruin work that I did by working with people who will show you how to do this wrong.

It's very simple logic, stop trying to make a problem to figure out something else for people who claim to be keeping people safe and can't.

mthoodlum, 2021-11-02:
You created the network configuration on the HowTo network. Now the devices on that network are erroneously being logged as offline even though they are online.

andrewhodel, 2021-11-02:
This entire thing is absurd and prefaced on the fact that the find command doesn't reliably work in routeros.

It's very difficult to be motivated about fixing nonsense that requires a working find command.

andrewhodel, 2021-11-02:
They are being logged as offline because they have been physically manipulated.

andrewhodel, 2021-11-02:
It is working on comcast networks on bzvpy, everywhere.

andrewhodel, 2021-11-02:
[image]

andrewhodel, 2021-11-02:
If someone walked into my house and modified the router, you have two choices... kill them or hold them in place until they are arrested and expect them to come back if not dead or injured.

That's reality.

Stop making up broken lies for me to fix your understanding of something.

closed 2021-11-02 by andrewhodel

andrewhodel, 2021-11-02:
Learn to accept freedom of speech if me listening to music drives people so crazy that they would do that.

Accept it, stop expecting me to fix it with sound.

andrewhodel, 2021-11-02:
This is simple human dynamics, people cannot just be stupid and expect to be taken care of... people that work get overwhelmed.

https://www.usatoday.com/story/news/world/2021/05/29/mass-grave-215-bodies-found-indigenous-school-canada/5262768001/

I don't know why this is something you want to know about, but you don't need to reflash radios to make up problems because I am only going to tell you truth of what will happen if property is that disrespected and people are that manipulated.

andrewhodel, 2021-11-02:
[image]

Comcast wan ip.