ispapp/ispapp-routeros-client #20: Controlled Access Point system Manager (CAPsMAN) wireless info

mthoodlum, 2021-11-16:
We have a client that configures every gateway with CAPsMAN so they can add an additional access point with minimal additional configuration. The received signal strength indicator and other wireless info isn't in the same menus if CAPsMAN is used.

andrewhodel, 2021-11-16:
https://wiki.mikrotik.com/wiki/Manual:CAPsMAN

```
/caps-man interface print
/caps-man registration-table print
```

andrewhodel, 2021-11-18:
> to @andrewhodel <https://github.com/andrewhodel>.

Send me temporary login credentials and the IP address for a device that
has a `/caps-man interface` entry that runs 802.11 and has connected
stations listed in `/caps-man registration-table`.

Andrew

closed 2021-11-23 by andrewhodel

mthoodlum, 2021-11-23:
[image]

commit 6213a3ca (2021-11-23): v1.50 added caps-man support, fixed #20