ispapp/ispapp-routeros-client #23: 100% CPU utilization

mthoodlum, 2021-11-24:
Devices with a lot of collector data hit 100% CPU utilization during data collection. This can be a particular problem when these devices go into in fast update mode. Typically a short spike in CPU usage every normal collector interval is not a problem. The CPU data collection should show the ambient CPU load not the CPU load caused by the data collection.

andrewhodel, 2021-11-24:
Run a thread that executes every ten seconds and averages the values at
each update interval.

Andrew

andrewhodel, 2021-11-24:
I was assuming it was using a few nanoseconds to calculate the load, it's already averaging it but Mikrotik doesn't document the interval.

You can see here the `pre-issue` script reports 10 to 15:

[image]

And this script reports 20 to 35:

[image]

It proves that for that measurement interval (undocumented) the `pre-issue` script is actually for such a short interval that it is reporting a lower cpu usage.

We know that for certain because the collectors script that executes that command runs every 60 seconds and the new script "avgCpuCollector" runs every 4 seconds.

andrewhodel, 2021-11-24:
It can be averaged easily with:

```
:set cpuLoad (([/system resource get cpu-load] + $cpuLoad) / 2);
```

This is the easiest way to do it, but truthfully Mikrotik should document if the `/system resource get cpu-load` command runs for a single cycle or averages a certain number.  The fact that the values are lower when ran in the collectors script is proof of malpractice.

andrewhodel, 2021-11-24:
fixed https://github.com/ispapp/ispapp-routeros-client/commit/b536afc39775d00ef9967c5174ec4e19bb3c0e33

closed 2021-11-24 by andrewhodel

mthoodlum, 2021-11-24:
I don't think we want a running average of CPU utilization. We want a real-time measurement. We don't want the process of collecting the CPU utilization to impact the value so we shouldn't get the value at the same time we do a bunch of other stuff.

andrewhodel, 2021-11-24:
You really don't understand what you are saying nor have you read the collector documentation.

```
System

collectors.system		object		object containing different system details

collectors.system.load		object		object containing system load values

load.one			number		one minute system load
load.five			number		five minute system load
load.fifteen			number		fifteen minute system load
load.processCount		number		total count of running processes
```

There is absolutely no purpose in collecting single cycle measurements, you are supposed to measure the processor temperature and ambient temperature and know the pre-tested input values for all processor usage rates then wrap the chip itself in a plastic case that has operating ranges for the environment (temperature, pressure, etc) but sub-atomic particles may change that.

You are dealing with a company that is writing code for a product that doesn't even document the cycle count (Mikrotik), what do you expect?

mthoodlum, 2021-11-24:
Okay! I understand now. Thanks.

andrewhodel, 2021-11-24:
Also it looks like an open winbox connection increases the system load to 40%, everything still remains in truth though.  Without documentation of the collection interval that their command uses and the low level C library access to run it fast enough to capture real data you cannot expect anything more than that.

I find it depressing that you wanted to go through their whole game of CPU collection intervals just to ask about sub-atomic particles and atmospheric pressure you fascist.