ispapp/ispapp-routeros-client #25: Email syntax changed in RouterOS v7

mthoodlum, 2022-01-14:
v6
/tool e-mail set start-tls=tls-only;

v7:
/tool e-mail set tls=starttls

andrewhodel, 2022-01-14:
starttls and tls-only are completely different regardless of the parameter key changing.

andrewhodel, 2022-01-14:
The routeros devices are again not following their programming.

The script has been set to run every 2 seconds and is not.

[image]
[image]
[image]

andrewhodel, 2022-01-14:
The lie is that the script doesn't start because of a problem with the email tool being setup wrong, but the email tool does not block forever as you can see in the /system/scripts/jobs entry.

andrewhodel, 2022-01-14:
Here is the bug Mikrotik must fix for this to work.

MikroTik support #[SUP-71609]: routeros script failure with email settings for v6 and v7

andrewhodel, 2022-04-03:
Workaround - https://github.com/ispapp/ispapp-routeros-client/commit/92ddedb7ca9883223510e0f70b947a0109a7c08e

closed 2022-04-03 by andrewhodel