ispapp/ispapp-routeros-client #34: send wifi information in config request

andrewhodel, 2022-05-19:
Only to be available when added as an unknown host.

This requires https://github.com/ispapp/collect-v2/issues/481 to be finished first because **setting wifi information will require resetting all existing router configuration**.

A routers existing wifi could be setup with an IPIP tunnel for example, the existing configuration needs to be reset to apply only the wifi SSID and key.

andrewhodel, 2022-06-11:
added to routeros client

closed 2022-06-11 by andrewhodel

mthoodlum, 2022-06-11:
Please also change the functionality to set the Wi-Fi settings on the physical interfaces, not the virtual interfaces. If we are passing previous Wi-Fi settings then we will avoid the problem of typos.

andrewhodel, 2022-06-11:
That's a terrible idea, it doesn't allow configuration with a single setting and should be done with a shell or with commands or with a console or with a terminal.

Making a UI for that or an interface is awful and constantly problematic and incapable of staying with things like SDR.

mthoodlum, 2022-06-11:
It allows bulk edits on that settings. It is no different than setting virtual interfaces, except it causes less problems.

andrewhodel, 2022-06-11:
You are wrong, you are misunderstanding that bulk commands will be added and that bulk edits are not possible because of the list of interfaces and requirement for multi step processes like creating bridges then adding interfaces to them.

A bulk tool allows commands, interfaces and UIs are not the right tool for that.

That's already been proven.

Stop acting like a fool because bulk commands are being worked on and you don't accept things until they are done in your logical arguments.