ispapp/ispapp-routeros-client #41: hAP AC^3 images not right

mthoodlum, 2022-07-20:
https://mikrotik.com/product/hap_ac3

Correct images:
RBD53iG-5HacD2HnD-1-3000x3000.webp
https://drive.google.com/drive/folders/1f0J4yCWSyLVmovcAaR9xOvWld6WUIMyD

/system routerboard print
       routerboard: yes
        board-name: hAP ac^3
             model: RBD53iG-5HacD2HnD
     serial-number: *********
     firmware-type: ipq4000
  factory-firmware: 6.47.10
  current-firmware: 6.47.10
  upgrade-firmware: 6.47.10

andrewhodel, 2022-07-21:
You mean to say the images are not being shown for this device, correct?

andrewhodel, 2022-07-21:
Then why did you not show a valid screenshot?

mthoodlum, 2022-07-21:
Because this is a public repo

andrewhodel, 2022-07-21:
That's not reasonable.

You can show the images and the device model without showing the host details.

andrewhodel, 2022-07-25:
Created in the correct repository - https://github.com/ispapp/collect-v2/issues/550

closed 2022-07-25 by andrewhodel