ispapp/ispapp-routeros-client #54: Improved error handling

mthoodlum, 2022-09-15:
Create a "debug" variable in the RouterOS script. That variable needs to be set to 1 by default and set to 0 after the device successfully sends a config request and an interval request. All the debugging print statements should have an "if debug = 1" in front.

There should also be an "error" variable that stores the data of the last failed request.

andrewhodel, 2022-09-18:
The only calls that can fail or not fail are the update fetch, the config fetch, the command response fetch and the send command response via SMTP call.

These commands all should output to the log on error by default.

Only these commands do not have that:

* command response fetch
* smtp command response fetch

The reason that they do not have error logging is because they repeat per the `/update` fetch that does have error output.

This issue has no purpose.

mthoodlum, 2022-09-18:
No, we need more error handling that is easy for customers to see and understand. We need printout statements during the install script that explain what data is being collected, sent, and if any errors are encounter then the errors need to be displayed.

To accomplish this there should be an ispapp-debug environmental variable that is set to 1 by default. After successful config and interval requests then the variable should be set to 0. Every print statement that is now commented out should simply have an if statement in front that says "if debug is 1 then print".

It is really important that our customers understand how our product works and how to troubleshoot and self-help.

@andrewhodel this issue is not assigned to you and it needs to stay open.

andrewhodel, 2022-10-28:
These problems were fixed by https://github.com/ispapp/ispapp-routeros-client/commit/36a537eff8d7f14faf1d4ccfbb9cbf6852f545db

andrewhodel, 2023-01-03:
Fixed in 2.23

https://github.com/ispapp/ispapp-routeros-client/commit/03c905dc4ffbe805629ecb8361a0932b956d4723

andrewhodel, 2023-02-25:
This was fixed on January 18th 2023 as shown in the commit log - https://github.com/ispapp/ispapp-routeros-client/commit/627462f74df1254385a22f19b203a0b9511a36bd

mthoodlum, 2023-04-04:
While debug1 please output to the router log as well.

andrewhodel, 2023-04-04:
https://github.com/ispapp/ispapp-routeros-client/commit/cf0df95cd163882253a53fa3429f09a41c550cd9

Already fixed.

closed 2023-04-04 by andrewhodel