ispapp/ispapp-routeros-client #58: Remove router/bridge settings

mthoodlum, 2022-10-21:
Remove these parts of the RouterOS client script:

virtual access points
hidden SSIDs
interface bridge ispapp-lan
interface bridge port
ip address 10.10.0.1/16
ip dhcp-server
ip dhcp-server network
ip pool ispapp-lan-pool
ip firewall nat masquerade

andrewhodel, 2022-10-25:
These are required to have a functional router that does not require configuration.

mthoodlum, 2022-11-01:
The ISPs are going to be loading their own ISP specific settings. ISPApp will just be used for performance monitoring, web shell, programmatically controlling devices using the server API, configuration management, and allowing the subscribers to set their own Wi-Fi network names and passwords. There is no reason for ISPApp to try to configure a router from scratch or change a router to a bridge or a bridge to a router.

Right now devices are in configuration loops constantly updating their settings.

Here is an example of /system history print

U nat rule added                     admin  write 
U dhcp server dhcp1 added            admin  write 
U dhcp network added                 admin  write 
U pool ispapp-lan-pool added         admin  write 
U address added                      admin  write 
U address removed                    admin  write 
U pool ispapp-lan-pool removed       admin  write 
U dhcp network removed               admin  write 
U dhcp server dhcp1 removed          admin  write 
U nat rule removed                   admin  write 

Here is an example of /system history print detail

 U redo=/ip firewall nat remove *1A69\r\n 
    undo=/ip firewall nat add action=masquerade chain=srcnat comment=\\r\n    
     ispapp-lan\r\n 
    action="nat rule removed" by="admin" policy=write 
    time=oct/28/2022 15:11:35 

 U redo=/system identity set name=lab-ROSv7-hAP-AC\r\n 
    undo=/system identity set name=lab-ROSv7-hAP-AC\r\n 
    action="system identity changed" by="admin" policy=write 
    time=oct/28/2022 15:11:35 

 U redo=/ip firewall nat add action=masquerade chain=srcnat comment=\\r\n    
     ispapp-lan\r\n 
    undo=/ip firewall nat remove *1A69\r\n action="nat rule added" by="admin" 
    policy=write time=oct/28/2022 15:11:21 

 U redo=/ip dhcp-server add address-pool=*7A0 disabled=no interface=\\r\n    
     ispapp-lan\r\n 
    undo=/ip dhcp-server remove *1A69\r\n action="dhcp server dhcp1 added" 
    by="admin" policy=write time=oct/28/2022 15:11:21 

 U redo=/ip dhcp-server network add address=10.10.0.0/16 dns-server=\\r\n    
     8.8.8.8,8.8.4.4 gateway=10.10.0.1\r\n 
    undo=/ip dhcp-server network remove *1A69\r\n action="dhcp network added" 
    by="admin" policy=write time=oct/28/2022 15:11:21 

 U redo=/ip pool add name=ispapp-lan-pool ranges=10.10.0.10-10.10.254.254\r\n 
    undo=/ip pool remove *7A0\r\n action="pool ispapp-lan-pool added" 
    by="admin" policy=write time=oct/28/2022 15:11:21 

 U redo=/ip address add address=10.10.0.1/16 interface=ispapp-lan\r\n 
    undo=/ip address remove *597F\r\n action="address added" by="admin" 
    policy=write time=oct/28/2022 15:11:21 

 U redo=/ip address remove *597E\r\n 
    undo=/ip address add address=10.10.0.1/16 disabled=no interface=\\r\n    
     ispapp-lan network=10.10.0.0\r\n 
    action="address removed" by="admin" policy=write 
    time=oct/28/2022 15:11:21 

 U redo=/ip pool remove ispapp-lan-pool\r\n 
    undo=/ip pool add name=ispapp-lan-pool ranges=10.10.0.10-10.10.254.254\r\n 
    action="pool ispapp-lan-pool removed" by="admin" policy=write 
    time=oct/28/2022 15:11:21 

 U redo=/ip dhcp-server network remove *1A68\r\n 
    undo=/ip dhcp-server network add address=10.10.0.0/16 caps-manager="" 
     \\r\n    dhcp-option="" dns-server=8.8.8.8,8.8.4.4 gateway=10.10.0.1 
     \\r\n    ntp-server="" wins-server=""\r\n 
    action="dhcp network removed" by="admin" policy=write 
    time=oct/28/2022 15:11:21 

 U redo=/ip dhcp-server remove *1A68\r\n 
    undo=/ip dhcp-server add address-pool=ispapp-lan-pool authoritative=yes 
     \\r\n    disabled=no interface=ispapp-lan lease-script="" lease-time=10m 
     \\r\n    name=dhcp1 use-radius=no\r\n 
    action="dhcp server dhcp1 removed" by="admin" policy=write 
    time=oct/28/2022 15:11:21 

 U redo=/ip firewall nat remove *1A68\r\n 
    undo=/ip firewall nat add action=masquerade chain=srcnat comment=\\r\n    
     ispapp-lan\r\n 
    action="nat rule removed" by="admin" policy=write 
    time=oct/28/2022 15:11:21 

 U redo=/system identity set name=lab-ROSv7-hAP-AC\r\n 
    undo=/system identity set name=lab-ROSv7-hAP-AC\r\n 
    action="system identity changed" by="admin" policy=write 
    time=oct/28/2022 15:11:21 

 U redo=/ip firewall nat add action=masquerade chain=srcnat comment=\\r\n    
     ispapp-lan\r\n 
    undo=/ip firewall nat remove *1A68\r\n action="nat rule added" by="admin" 
    policy=write time=oct/28/2022 15:11:05 

 U redo=/ip dhcp-server add address-pool=ispapp-lan-pool disabled=no \\r\n    
     interface=ispapp-lan\r\n 
    undo=/ip dhcp-server remove *1A68\r\n action="dhcp server dhcp1 added" 
    by="admin" policy=write time=oct/28/2022 15:11:05

andrewhodel, 2022-11-02:
What bridge are the wifi networks that subscribers create supposed to use?

mthoodlum, 2022-11-02:
That is why ISPApp should only modify the SSID and encryption key for wlan1 and wlan2 which are already going to be on the correct bridges as per the standardized configurations of the ISPs. Then the subscribers can change the WI-Fi network names and passwords without even worrying about bridge ports or any of these other settings.

mthoodlum, 2022-11-30:
To be clear, the ISPApp RouterOS client needs to stop adding and removing configurations for the following:

virtual access points
hidden SSIDs
interface bridge
interface bridge port
ip address
ip dhcp-server
ip dhcp-server network
ip pool
ip firewall nat masquerade
ipv6

Existing customers who do not use those settings can remove them from their devices using bulk commands. Existing customers who are using those settings can continue using those settings, or they can change those settings using bulk commands.

commit 9391eada (2022-12-01): added easy to understand ispappRemoveConfiguration script, simplified ispappConfig script, fixed #58, added initial/structural support for #55 (waiting on RouterOS issue SUP-99759) and #31
closed 2022-12-01 by andrewhodel