ispapp/ispapp-routeros-client #63: sending negative ping data

andrewhodel, 2023-01-03:
[image]

andrewhodel, 2023-02-17:
The ping command not resolving an address causes this.

closed 2023-02-17 by andrewhodel

arvinalucitain, 2023-06-17:
I've seen some issues that after adding another ping host address, that address doesn't disappear on the dashboard of ISPApp.
[image]

arvinalucitain, 2023-06-22:
Ping result update for the demo device.
[image]